birkenfeld/ads-rs #7: Reading / writing strings of length 255

Mebus, 2022-06-21:
Hey Georg,

I am trying to read / write strings, that are 255 characters long from the PLC. I tried it like this, but this limits me to 32 bytes:

```
let handle = ads::Handle::new(device, "MyValue").unwrap();
let mut somevalue: [u8; 32];
somevalue =  handle.read_value().unwrap();
```

What is the correct way to achieve my goal?

Thanks

Mebus

birkenfeld, 2022-06-21:
You have two options:

* since you read a byte array anyway, use `Handle::read` instead of `read_value`
* create a type with the correct length with `ads::make_string_type!` and use that

Mebus, 2022-06-21:
This seems to work:

```
    const LEN: usize = 255;
    let mut msg: [u8; LEN] = [0; LEN];
    handle.read(&mut msg).unwrap();
    println!("MY_SYMBOL value is {:?}", msg);
    let thestring = str::from_utf8(&msg).unwrap();
    println!("{}", thestring);
```

:-)

Mebus

birkenfeld, 2022-06-21:
Yes, that's option 1. Option 2, to compare:
```
ads::make_string_type!(String255, 255);  // on toplevel

let plc_str: String255 = handle.read_value().unwrap();
let rust_str: String = plc_str.try_into().unwrap();
```
You might have to `use std::convert::TryInto` if you're using Rust edition 2015/18.

closed 2022-08-26 by birkenfeld