birkenfeld/ads-rs #8: ads::symbol::get_symbol_info breaks on large TwinCAT projects

alex-boring, 2022-09-20:
Hey,
great library, thanks a lot for that!

I am running into an issue when using `ads::symbol::get_symbol_info` on a large TwinCAT project. With a small one it works just fine (tested with a couple dozen variables). Not sure where the threshold is when it breaks.

Any help would be appreciated!
Thanks,
Alex

Output: 
```
thread 'main' panicked at 'range end index 1993 out of range for slice of length 1024', library\core\src\slice\index.rs:73:5
stack backtrace:
   0:     0x7ff75966713f - std::backtrace_rs::backtrace::dbghelp::trace
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\..\..\backtrace\src\backtrace\dbghelp.rs:98
   1:     0x7ff75966713f - std::backtrace_rs::backtrace::trace_unsynchronized
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
   2:     0x7ff75966713f - std::sys_common::backtrace::_print_fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:66
   3:     0x7ff75966713f - std::sys_common::backtrace::_print::impl$0::fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:45
   4:     0x7ff7596780ea - core::fmt::write
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\fmt\mod.rs:1197
   5:     0x7ff7596633c9 - std::io::Write::write_fmt<std::sys::windows::stdio::Stderr>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\io\mod.rs:1672
   6:     0x7ff7596697cb - std::sys_common::backtrace::_print
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:48
   7:     0x7ff7596697cb - std::sys_common::backtrace::print
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:35
   8:     0x7ff7596697cb - std::panicking::default_hook::closure$1
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:295
   9:     0x7ff75966944b - std::panicking::default_hook
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:314
  10:     0x7ff759669d79 - std::panicking::rust_panic_with_hook
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:698
  11:     0x7ff759669c7d - std::panicking::begin_panic_handler::closure$0
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:588
  12:     0x7ff759667ac7 - std::sys_common::backtrace::__rust_end_short_backtrace<std::panicking::begin_panic_handler::closure_env$0,never$>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:138
  13:     0x7ff759669959 - std::panicking::begin_panic_handler
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:584
  14:     0x7ff75967f495 - core::panicking::panic_fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\panicking.rs:142
  15:     0x7ff7596799cb - core::slice::index::slice_end_index_len_fail_rt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\slice\index.rs:73
  16:     0x7ff759676b19 - core::ops::function::FnOnce::call_once<never$ (*)(usize,usize),tuple$<usize,usize> >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\ops\function.rs:248
  17:     0x7ff759676d79 - core::intrinsics::const_eval_select<tuple$<usize,usize>,never$ (*)(usize,usize),never$ (*)(usize,usize),never$>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\intrinsics.rs:2376
  18:     0x7ff75967f559 - core::slice::index::slice_end_index_len_fail
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\slice\index.rs:67
  19:     0x7ff75962308e - core::slice::index::impl$3::index_mut<u8>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\slice\index.rs:315
  20:     0x7ff75962327e - core::slice::index::impl$4::index_mut<u8>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\slice\index.rs:356
  21:     0x7ff75960cb58 - core::slice::index::impl$1::index_mut<u8,core::ops::range::RangeTo<usize> >    
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\slice\index.rs:30
  22:     0x7ff75960dd6e - core::array::impl$16::index_mut<u8,core::ops::range::RangeTo<usize>,1024>      
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\array\mod.rs:298
  23:     0x7ff759621c6b - ads::symbol::decode_symbol_info::decode_type_info
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\ads-0.4.1\src\symbol.rs:252
  24:     0x7ff75961f828 - ads::symbol::decode_symbol_info
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\ads-0.4.1\src\symbol.rs:296
  25:     0x7ff75961f418 - ads::symbol::get_symbol_info
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\ads-0.4.1\src\symbol.rs:213
  26:     0x7ff7595eb025 - rust_config_param_viewer::get_ads_symbols
                               at C:\projects\rust_config_param_viewer\src\main.rs:30
  27:     0x7ff7595eb34e - rust_config_param_viewer::main
                               at C:\projects\rust_config_param_viewer\src\main.rs:48
  28:     0x7ff7595ec41b - core::ops::function::FnOnce::call_once<void (*)(),tuple$<> >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\ops\function.rs:248
  29:     0x7ff7595e32ab - std::sys_common::backtrace::__rust_begin_short_backtrace<void (*)(),tuple$<> > 
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\std\src\sys_common\backtrace.rs:122
  30:     0x7ff7595e49b1 - std::rt::lang_start::closure$0<tuple$<> >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\std\src\rt.rs:145
  31:     0x7ff75965f2c1 - core::ops::function::impls::impl$2::call_once
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\ops\function.rs:280
  32:     0x7ff75965f2c1 - std::panicking::try::do_call
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:492
  33:     0x7ff75965f2c1 - std::panicking::try
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:456
  34:     0x7ff75965f2c1 - std::panic::catch_unwind
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panic.rs:137
  35:     0x7ff75965f2c1 - std::rt::lang_start_internal::closure$2
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\rt.rs:128
  36:     0x7ff75965f2c1 - std::panicking::try::do_call
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:492
  37:     0x7ff75965f2c1 - std::panicking::try
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:456
  38:     0x7ff75965f2c1 - std::panic::catch_unwind
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panic.rs:137
  39:     0x7ff75965f2c1 - std::rt::lang_start_internal
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\rt.rs:128
  40:     0x7ff7595e497f - std::rt::lang_start<tuple$<> >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\std\src\rt.rs:144
  41:     0x7ff7595eb376 - main
  42:     0x7ff75967dba0 - invoke_main
                               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78  
  43:     0x7ff75967dba0 - __scrt_common_main_seh
                               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288 
  44:     0x7ff974a47034 - BaseThreadInitThunk
  45:     0x7ff9760a26a1 - RtlUserThreadStart
error: process didn't exit successfully: `target\debug\rust_config_param_viewer.exe` (exit code: 101) ```

birkenfeld, 2022-09-21:
Thanks for the report. Looks like a symbol's comment is longer than 1024 bytes, which is fair enough. (The comment isn't currently used anyway...)

alex-boring, 2022-09-21:
Thanks for the fast answer! How would you recommend working around that? (I can't change the comment length in the PLC code in this use case).

birkenfeld, 2022-09-21:
Should be fixed in master, can you try with it?

closed 2022-09-21 by birkenfeld

alex-boring, 2022-09-21:
Thanks, still the same issue with any large projects. I am not sure if the comment length is the cause. I tested it on a small project with a couple variables and a long comment behind it with 4500 characters. No problem with the long comment on the small project.

Code used for testing:
```
PROGRAM MAIN
VAR
	heartbeat : USINT;
	test_int : INT := 42;
	test_float : REAL := 3.14159;
	test_string : STRING(20) := 'This is a string';
	test_double : LREAL := 423.14159;
	test_bool : BOOL := TRUE;
	test_array : ARRAY[0..9] OF INT := [1,-101,10101,123,-6665,5412,2222,1234,5667];	// adsffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff ... this goes on for another 4000 characters

END_VAR
```

birkenfeld, 2022-09-21:
Are you sure the traceback is the same? Maybe there's still a related bug...

I assume the symbol/type info contains confidential info and you can't share it?

alex-boring, 2022-09-21:
I am running into problems building the master from source right now using cargo. Will fix this and get back to you. Thanks!

birkenfeld, 2022-09-21:
I also checked how comments end up in the symbol info now, and the only way I could produce a non-empty comment field was to place the comment on a member of a `STRUCT` and then to add an instance of that struct to e.g. `VAR` in `PROGRAM MAIN`.

Placed directly on a `VAR` item it would not show up...

alex-boring, 2022-09-21:
Testing with `STRUCT` on the tag 0.4.1 gets me the error message with any comment, no matter how long. Can't build from Master right now, but if it works for you with `STRUCT` comments I hope this solves my problem in your next tag.

Used for testing on 0.4.1 which fails, struct has one instance in MAIN:
```
// 
TYPE DUT :
STRUCT	// 
	test : INT; // 
END_STRUCT
END_TYPE

```

birkenfeld, 2022-09-21:
Anyway, with such short comments I see no failure in 0.4.1 either. I'm a bit confused by what's going on - can you make sure the traceback still mentions `'range end index xxx out of range for slice of length 1024'` and   ``C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\ads-0.4.1\src\symbol.rs:252`?

If yes, the other question is if there is an issue activating the project after updating, so that you are maybe still seeing the old instance with the long comment...

alex-boring, 2022-09-21:
Thanks, I didn't know the command to pull from git directly. Using `ads = { git = "https://github.com/birkenfeld/ads-rs.git" }` in the `Cargo.toml` file and the same large project I am getting a new error message, and it is printing my first lines (just for my test setup):

```
Starting Execution at time 2022-09-21T16:45:28.716185300-05:00.
[2022-09-21T21:45:28Z INFO  actix_server::builder] Starting 8 workers
[2022-09-21T21:45:28Z INFO  actix_server::server] Actix runtime found; starting in Actix runtime
"GVL_Constants.NDX_MAX_XXXX" type: "USINT" value: 4
"MAIN._M_Test.XXXX" type: "ULINT" value: 0
"MAIN._M_Test.XYZ type: "ULINT" value: 0
"MAIN._M_Test.YD" type: "ULINT" value: 5
"MAIN._M_Test.ABC" type: "ULINT" value: 0
"MAIN._M_Test.EFG" type: "BOOL" value: false
"MAIN._M_Test.HIJ" type: "ULINT" value: 0
"MAIN.ActiveXXXX" type: "REFERENCE TO FB_XXXX" value: [123, 1, 1, 1, 1, 1, 1, 255]"MAIN.ActiveXXXXX" type: "E_YYYYY" value: [0, 0, 0, 0]
thread 'actix-rt|system:0|arbiter:0' panicked at 'Failed to read the data from the PLC handle with error read data: Reading/writing not permitted (0x704)', src\resource\ads_communication.rs:38:31
stack backtrace:
   0:     0x7ff69ef4d3cf - std::backtrace_rs::backtrace::dbghelp::trace
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\..\..\backtrace\src\backtrace\dbghelp.rs:98
   1:     0x7ff69ef4d3cf - std::backtrace_rs::backtrace::trace_unsynchronized
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
   2:     0x7ff69ef4d3cf - std::sys_common::backtrace::_print_fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:66
   3:     0x7ff69ef4d3cf - std::sys_common::backtrace::_print::impl$0::fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:45
   4:     0x7ff69ef6429a - core::fmt::write
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\fmt\mod.rs:1197
   5:     0x7ff69ef47619 - std::io::Write::write_fmt<std::sys::windows::stdio::Stderr>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\io\mod.rs:1672
   6:     0x7ff69ef4fa5b - std::sys_common::backtrace::_print
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:48
   7:     0x7ff69ef4fa5b - std::sys_common::backtrace::print
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:35
   8:     0x7ff69ef4fa5b - std::panicking::default_hook::closure$1
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:295
   9:     0x7ff69ef4f6db - std::panicking::default_hook
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:314
  10:     0x7ff69ef50009 - std::panicking::rust_panic_with_hook
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:698
  11:     0x7ff69ef4ff0d - std::panicking::begin_panic_handler::closure$0
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:588
  12:     0x7ff69ef4dd57 - std::sys_common::backtrace::__rust_end_short_backtrace<std::panicking::begin_panic_handler::closure_env$0,never$>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:138
  13:     0x7ff69ef4fbe9 - std::panicking::begin_panic_handler
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:584
  14:     0x7ff69efbee15 - core::panicking::panic_fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\panicking.rs:142
  15:     0x7ff69e8d6699 - rust_config_param_viewer::api::ads_communication::get_ads_symbols
                               at C:\projects\rust_config_param_viewer\src\resource\ads_communication.rs:38
  16:     0x7ff69e96de3c - rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn$0  
                               at C:\projects\rust_config_param_viewer\src\resource\api.rs:58
  17:     0x7ff69e8d38c7 - core::future::from_generator::impl$1::poll<enum$<rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn_env$0> >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\future\mod.rs:91
  18:     0x7ff69e96697e - actix_web::handler::handler_service::closure$0::async_block$0<core::future::from_generator::GenFuture<enum$<rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn_env$0> > (*)(),tuple$<> >
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\actix-web-4.2.1\src\handler.rs:105
  19:     0x7ff69e8d30a7 - core::future::from_generator::impl$1::poll<enum$<actix_web::handler::handler_service::closure$0::async_block_env$0<core::future::from_generator::GenFuture<enum$<rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn_env$0> > (*)(),tup
[... cut ...]
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\alloc\src\boxed.rs:1951
  92:     0x7ff69ef5295c - alloc::boxed::impl$44::call_once
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\alloc\src\boxed.rs:1951
  93:     0x7ff69ef5295c - std::sys::windows::thread::impl$0::new::thread_start
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys\windows\thread.rs:56
  94:     0x7ff974a47034 - BaseThreadInitThunk
  95:     0x7ff9760a26a1 - RtlUserThreadStart

```

alex-boring, 2022-09-22:
@birkenfeld Do you have any example code for getting a struct with its fields (Struct inside of Struct inside of Struct) using `get_symbol_info()`? I need to get the structure of a specific struct in the PLC code including symbol names, so that I can pull the values for each symbol using handle.read

Sorry for posting it here, did not want to open a new issue for a question

birkenfeld, 2022-09-22:
Have you looked at `examples/adstool.rs`? Its `var list` subcommand should do something close to that.

commit d987790e (2022-09-21): symbol: fix for long (>1024b) comments

fixes #8