birkenfeld/ads-rs #9: "Failed to read data" from PLC

birkenfeld, 2022-09-22:
Using `ads = { git = "https://github.com/birkenfeld/ads-rs.git" }` in the `Cargo.toml` file and the same large project I am getting a new error message, and it is printing my first lines (just for my test setup). Would you like me to open a new Issue for that error message?

```
Starting Execution at time 2022-09-21T16:45:28.716185300-05:00.
[2022-09-21T21:45:28Z INFO  actix_server::builder] Starting 8 workers
[2022-09-21T21:45:28Z INFO  actix_server::server] Actix runtime found; starting in Actix runtime
"GVL_Constants.NDX_MAX_XXXX" type: "USINT" value: 4
"MAIN._M_Test.XXXX" type: "ULINT" value: 0
"MAIN._M_Test.XYZ type: "ULINT" value: 0
"MAIN._M_Test.YD" type: "ULINT" value: 5
"MAIN._M_Test.ABC" type: "ULINT" value: 0
"MAIN._M_Test.EFG" type: "BOOL" value: false
"MAIN._M_Test.HIJ" type: "ULINT" value: 0
"MAIN.ActiveXXXX" type: "REFERENCE TO FB_XXXX" value: [123, 1, 1, 1, 1, 1, 1, 255]"MAIN.ActiveXXXXX" type: "E_YYYYY" value: [0, 0, 0, 0]
thread 'actix-rt|system:0|arbiter:0' panicked at 'Failed to read the data from the PLC handle with error read data: Reading/writing not permitted (0x704)', src\resource\ads_communication.rs:38:31
stack backtrace:
   0:     0x7ff69ef4d3cf - std::backtrace_rs::backtrace::dbghelp::trace
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\..\..\backtrace\src\backtrace\dbghelp.rs:98
   1:     0x7ff69ef4d3cf - std::backtrace_rs::backtrace::trace_unsynchronized
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\..\..\backtrace\src\backtrace\mod.rs:66
   2:     0x7ff69ef4d3cf - std::sys_common::backtrace::_print_fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:66
   3:     0x7ff69ef4d3cf - std::sys_common::backtrace::_print::impl$0::fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:45
   4:     0x7ff69ef6429a - core::fmt::write
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\fmt\mod.rs:1197
   5:     0x7ff69ef47619 - std::io::Write::write_fmt<std::sys::windows::stdio::Stderr>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\io\mod.rs:1672
   6:     0x7ff69ef4fa5b - std::sys_common::backtrace::_print
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:48
   7:     0x7ff69ef4fa5b - std::sys_common::backtrace::print
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:35
   8:     0x7ff69ef4fa5b - std::panicking::default_hook::closure$1
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:295
   9:     0x7ff69ef4f6db - std::panicking::default_hook
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:314
  10:     0x7ff69ef50009 - std::panicking::rust_panic_with_hook
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:698
  11:     0x7ff69ef4ff0d - std::panicking::begin_panic_handler::closure$0
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:588
  12:     0x7ff69ef4dd57 - std::sys_common::backtrace::__rust_end_short_backtrace<std::panicking::begin_panic_handler::closure_env$0,never$>
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys_common\backtrace.rs:138
  13:     0x7ff69ef4fbe9 - std::panicking::begin_panic_handler
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\panicking.rs:584
  14:     0x7ff69efbee15 - core::panicking::panic_fmt
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\core\src\panicking.rs:142
  15:     0x7ff69e8d6699 - rust_config_param_viewer::api::ads_communication::get_ads_symbols
                               at C:\projects\rust_config_param_viewer\src\resource\ads_communication.rs:38
  16:     0x7ff69e96de3c - rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn$0  
                               at C:\projects\rust_config_param_viewer\src\resource\api.rs:58
  17:     0x7ff69e8d38c7 - core::future::from_generator::impl$1::poll<enum$<rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn_env$0> >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\future\mod.rs:91
  18:     0x7ff69e96697e - actix_web::handler::handler_service::closure$0::async_block$0<core::future::from_generator::GenFuture<enum$<rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn_env$0> > (*)(),tuple$<> >
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\actix-web-4.2.1\src\handler.rs:105
  19:     0x7ff69e8d30a7 - core::future::from_generator::impl$1::poll<enum$<actix_web::handler::handler_service::closure$0::async_block_env$0<core::future::from_generator::GenFuture<enum$<rust_config_param_viewer::api::impl$4::register::get_sysParamList::async_fn_env$0> > (*)(),tup
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\future\mod.rs:91
  20:     0x7ff69e9fea49 - core::future::future::impl$1::poll<alloc::boxed::Box<dyn$<core::future::future::Future<assoc$<Output,enum$<core::result::Result<actix_web::service::ServiceResponse<actix_http::body::boxed::BoxBody>,actix_web::error::error::Error> > > > >,alloc::alloc::Glo
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\future\future.rs:124
  21:     0x7ff69ea0d43d - actix_web::resource::impl$2::register::closure$0::async_block$0<actix_web::resource::ResourceEndpoint,actix_http::body::boxed::BoxBody>
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\actix-web-4.2.1\src\resource.rs:383
  22:     0x7ff69e9f2a17 - core::future::from_generator::impl$1::poll<enum$<actix_web::resource::impl$2::register::closure$0::async_block_env$0<actix_web::resource::ResourceEndpoint,actix_http::body::boxed::BoxBody> > >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\future\mod.rs:91
  23:     0x7ff69e9fea49 - core::future::future::impl$1::poll<alloc::boxed::Box<dyn$<core::future::future::Future<assoc$<Output,enum$<core::result::Result<actix_web::service::ServiceResponse<actix_http::body::boxed::BoxBody>,actix_web::error::error::Error> > > > >,alloc::alloc::Glo
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\future\future.rs:124
  24:     0x7ff69e8d3f80 - actix_web::middleware::logger::impl$4::poll<actix_web::app_service::AppRouting,actix_http::body::boxed::BoxBody>
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\actix-web-4.2.1\src\middleware\logger.rs:302
  25:     0x7ff69e921c56 - actix_service::map_err::impl$4::poll<actix_web::app_service::AppInitService<actix_web::middleware::logger::LoggerMiddleware<actix_web::app_service::AppRouting>,actix_web::middleware::logger::StreamLog<actix_http::body::boxed::BoxBody> >,actix_http::reques
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\actix-service-2.0.2\src\map_err.rs:99
  26:     0x7ff69e8fbe28 - actix_http::h1::dispatcher::InnerDispatcher<tokio::net::tcp::stream::TcpStream,actix_service::map_err::MapErr<actix_web::app_service::AppInitService<actix_web::middleware::logger::LoggerMiddleware<actix_web::app_service::AppRouting>,actix_web::middleware:
                               at C:\Users\testUser\.cargo\registry\src\github.com-1ecc6299db9ec823\actix-http-3.2.2\src\h1\dispatcher.rs:653
[... 112 lines cut ...]
  84:     0x7ff69ea4fa64 - core::panic::unwind_safe::impl$23::call_once<tuple$<>,std::thread::impl$0::spawn_unchecked_::closure$1::closure_env$0<actix_rt::arbiter::impl$2::with_tokio_rt::closure_env$0<actix_server::worker::impl$10::start::closure_env$1>,tuple$<> > >
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f\library\core\src\panic\unwind_safe.rs:271
[... cut ...]
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\alloc\src\boxed.rs:1951
  93:     0x7ff69ef5295c - std::sys::windows::thread::impl$0::new::thread_start
                               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library\std\src\sys\windows\thread.rs:56
  94:     0x7ff974a47034 - BaseThreadInitThunk
  95:     0x7ff9760a26a1 - RtlUserThreadStart

```

_Originally posted by @alex-boring in https://github.com/birkenfeld/ads-rs/issues/8#issuecomment-1254264293_

birkenfeld, 2022-09-22:
So this is a valid error from the PLC (as opposed to an ads-rs internal bug in #8).

It will be hard to debug without knowing what exactly is requested there. Can you share the relevant part of the code (the immediate cause is in your `get_ads_symbols` function)?

It might also just be that the PLC disallows reading the value of certain symbols. In that case you'll have to catch that error and handle it instead of panicking by calling `unwrap` or similar...

birkenfeld, 2022-09-22:
(cc @alex-boring if you weren't subscribed automatically)

alex-boring, 2022-09-22:
That error is on me, I fixed the issue on my side. Thanks for the hints. You can close this one out!

birkenfeld, 2022-09-22:
Great! I'll release the current master as 0.4.2 then.

closed 2022-09-22 by birkenfeld