birkenfeld/ads-rs #10: Decoding and flattening of struct array - question

alex-boring, 2022-10-10:
Hey, 
the library is working great so far!

One question I have when using it that you might be able to answer: do you have an example on how to decode (and flatten) arrays, especially struct arrays using the library?

Example Struct:
```
TYPE ST_Test:
STRUCT
	IN	: ST_Test_IO_IN;
	OUT	: ST_Test_IO_OUT;
        Test    : LREAL;
END_STRUCT
END_TYPE
```

Example Usage in MAIN Program:
```
Test	: ARRAY[0..9] OF ST_Test;
```

Now when I am using the Rust app to get all the information on the symbols in a struct it would be great to get a flattened array like this that I can work with and forward (without needing to know anything else about the PLC code and the structure of the struct). How the print could ideally look like:

```
Test[0].IN.whatever = 'some string'
Test[0].OUT.someLightOn = True
Test[0].Test = 1
.
.
.
Test[9].IN.whatever = 'some other string'
Test[9].OUT.someLightOn = False
Test[9].Test = 9
```

This becomes even trickier for multi dimensional arrays I assume.

Getting the nested symbol information for non-array symbols works like a charm with your recursive function in the examples.

Thanks in advance,
Alex

birkenfeld, 2022-10-11:
If I understand correctly, you need to look at the `array` member of the type info struct to get the lower and upper bounds. Then you can go through each index and recurse through the members.

birkenfeld, 2022-11-18:
I'm closing this for now, feel free to reopen if you are not satisfied.

closed 2022-11-18 by birkenfeld