birkenfeld/ads-rs #12: Making ads::client::Client implement the Send and Sync traits by not using RefCell and Cell in it

andrewraytan, 2023-04-13:
Hi, this is a great library so far.

I'm trying to share a single Device object across several different threads so I can make ADS requests using just 1 connection. However, because Device contains Client which uses RefCell and Cell, it doesn't implement Send and Sync, and therefore cannot be sent across different threads.

This is also an issue when trying to use the Device within a tokio asynchronous method, since calling await on a tokio runtime involves possibly moving the runtime to different threads, requiring the Device to implement Send and Sync.

Thanks,
Andrew

birkenfeld, 2023-04-14:
Hi, thanks for raising this issue!

While it wouldn't be hard to make the Client `Send`, the current implementation is not compatible with making requests simultaneously from multiple threads (which would be required for `Sync`). Such an implementation could be made, but currently I don't have enough time for it.

If you would like to work on it, I can give pointers.

andrewraytan, 2023-04-17:
Sure, I may do it at some point, please give these pointers!