birkenfeld/ads-rs #16: Compilation currently broken

gimmi, 2023-05-04:
Hello, I am triying to run the example but seems that the build is broken:

```
> git clone https://github.com/birkenfeld/ads-rs.git
> cd ads-rs
> cargo build
error: failed to download `syn v2.0.15`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `C:\Users\gimmi\.cargo\registry\src\github.com-1ecc6299db9ec823\syn-2.0.15\Cargo.toml`

Caused by:
  failed to parse the `edition` key

Caused by:
  this version of Cargo is older than the `2021` edition, and only supports `2015` and `2018` editions.
---

birkenfeld, 2023-05-04:
Thanks, fixed by pinning a dependency.

I'll probably release a new minor version soon, it might get higher MSRV.

closed 2023-05-04 by birkenfeld

gimmi, 2023-05-04:
Wow that was quick!

Confirm compiling fine now.

commit c08d86de (2023-05-04): Fix thiserror to avoid syn-2 dependency.

fixes #16