birkenfeld/ads-rs #17: Unable to set TC status

gimmi, 2023-05-05:
Hello, 

I am trying to use the `adstool` example to set device status, but I am unable to do it.

The same operation works fine using the TwinCAT Powershell API

[image]

birkenfeld, 2023-05-05:
Also, there is `Reconfig` which must apparently be used to set to config mode, instead of `Config`.

gimmi, 2023-05-05:
Interesting, will try later

gimmi, 2023-05-06:
Confirm working, here is the full test

Starting from state `Config`

```
C:\temp>adstool.exe 127.0.0.1/192.168.15.4.1.1 state
Device: TwinCAT System 3.1.4024
Current state: Config
```

Go from `Config` to `Run`

```
C:\temp>adstool.exe 127.0.0.1/192.168.15.4.1.1 state Reset
Device: TwinCAT System 3.1.4024
Current state: Config
Set new state: Reset
```

Go from `Run` to `Run` (restart)

```
C:\temp>adstool.exe 127.0.0.1/192.168.15.4.1.1 state Reset
Device: TwinCAT System 3.1.4024
Current state: Run
Set new state: Reset
```

Go from `Run` to `Config`

```
C:\temp>adstool.exe 127.0.0.1/192.168.15.4.1.1 state Reconfig
Device: TwinCAT System 3.1.4024
Current state: Run
Set new state: Reconfig
```

Go from `Config` to `Config`

```
C:\temp>adstool.exe 127.0.0.1/192.168.15.4.1.1 state Reconfig
Device: TwinCAT System 3.1.4024
Current state: Config
Set new state: Reconfig
```

birkenfeld, 2023-05-06:
Great! I'll add some text to the help of the `state` command, but I don't want to change the behavior as long as I don't have full knowledge of what Beckhoff are doing, and I don't think it's documented. Ok?

gimmi, 2023-05-06:
absolutely, I think your driver is doing the right thing by exposing ADS api as is, then the fact that ADS protocol has strange quirks here and there is something that need to be handled by the application code

commit dbe4c281 (2023-05-06): adstool: add a note about state transitions

fixes #17
closed 2023-05-06 by birkenfeld