birkenfeld/ads-rs #18: silent panic if twincat restarts mid-read

chammel4, 2023-06-13:
I'm finding that if a twincat restart completes (not sure yet exactly at which state transition the problem occurs) during either device.get_state() or handle.read(), the call never returns with a Result<> - this leads me to believe something in the background silently panics.

Is there something you're aware of that can be monitored to explicitly avoid this scenario? or otherwise, suggestions for where this silent panic may be taking place?

Thanks!

birkenfeld, 2023-06-13:
Are you creating the client with `Timeouts::none()`? It is recommended to use a timeout to avoid this.

chammel4, 2023-06-13:
good call, kind sir! That took care of it - now nicely riding through TC restarts.  I didn't realize I had used Timeouts::none

chammel4, 2023-06-13:
nothing to be done - Timeouts::none was erroneously used

closed 2023-06-13 by chammel4