birkenfeld/ads-rs #23: Notifications for named symbols possible?

Mebus, 2024-04-07:
Hallo,

is it possible to add notifications for named symbols, so I don't have to know the index group and index offset?

Mebus

birkenfeld, 2024-04-07:
Hm, good question. At least only the way with group/offset is documented on the Beckhoff pages, but knowing Beckhoff, there is some undocumented access to do it.

Mebus, 2024-04-09:
Where can I get the group / offset from? The tmc file?

Mebus

keenanjohnson, 2024-08-08:
Hey @Mebus there isn't a great example, but you need to do a couple of things:

1. First lookup the group / offset using the symbol::[get_location](https://github.com/birkenfeld/ads-rs/blob/7d4ca8a81f86e2fac14432c39df9ec20d352ab99/src/symbol.rs#L81) method:
2. Then you can subscribe to a notifcation for the named symbol

It's not one function but you can do as I did and easily write a little wrapper function that will do both for you.

**Get Location**

```
let symbol_location = match ads::symbol::get_location(dev, "MAIN.BOOL2") {
    Ok(location) => location,
    Err(e) => {
        eprintln!("Failed to get location: {}", e);
        return;
    }
};
```

symbol_location will then be a tuple like (0xF030, 0x2107) where the first term is the gorup and the second is the offset.

You can then pass those to the add_notification function: https://github.com/birkenfeld/ads-rs/blob/7d4ca8a81f86e2fac14432c39df9ec20d352ab99/src/client.rs#L786

closed 2024-08-09 by birkenfeld