birkenfeld/ads-rs #26: Enable Github Actions

keenanjohnson, 2024-08-08:
hey @birkenfeld !

I just started using this project as an alternative to the official Beckhoff C++ lib and I love it!

I would really like to see this project succeed in the long term, so just poking around to see what I can do to help improve the stability / confidence.

I noticed that although there is a github actions workflow defined, it doesn't look like it is running. Get that up and running seems like an obvious first step to increase the quality / stability of the lib.

Is there any reason you haven't enabled the workflow and anything I can do to help there?

birkenfeld, 2024-08-09:
Hi @keenanjohnson, thanks for the nice words!

I checked the workflow and it looked fine, but I updated it nevertheless and added a few bells and whistles :)
In https://github.com/birkenfeld/ads-rs/pull/27 everything ran as expected.

I'll be happy for other suggestions, and of course, also for code PRs if you have something to submit :+1:

keenanjohnson, 2024-08-09:
Excellent! Yes I see that it's running now. I guess that it had just been too long since a commit and Github stopped running the CI workflow? 

I'm sure I'll be submitting a few things into the project. One other small suggestion I might make for Github is to turn on the "Discussions" feature if you want to give the community a place to ask questions rather than opening issues for all questions which I've seen be quite helpful for many projects.

Thanks!

closed 2024-08-09 by keenanjohnson

birkenfeld, 2024-08-09:
Discussions are enabled now!

keenanjohnson, 2024-08-09:
Excellent! Thanks for the project!