birkenfeld/ads-rs #29: Error: Ads("read state", "Port disabled - system service not started", 18)

yichaoliang, 2024-10-14:
Hi,

I was trying to create a rust program to talk to a twinCAT RT running on the same PC.

Here's the code:
`// Open a connection to an ADS device identified by hostname/IP and port.
// For TwinCAT devices, a route must be set to allow the client to connect.
// The source AMS address is automatically generated from the local IP,
// but can be explicitly specified as the third argument.
// use ads::Result;
pub use ads::client::AdsState;
pub use ads::client::Client;
pub use ads::client::Device;
pub use ads::client::Source;
pub use ads::client::Timeouts;
pub use ads::errors::Error;
pub use ads::errors::Result;
pub use ads::file::File;
pub use ads::netid::AmsAddr;
pub use ads::netid::AmsNetId;
pub use ads::netid::AmsPort;
pub use ads::symbol::Handle;

fn main() -> ads::Result<()> {
    // Open a connection to an ADS device identified by hostname/IP and port.
    // For TwinCAT devices, a route must be set to allow the client to connect.
    // The source AMS address is automatically generated from the local IP,
    // but can be explicitly specified as the third argument.
    let client = ads::Client::new(("127.0.0.1", ads::PORT), ads::Timeouts::none(),
                                  ads::Source::Auto)?;

    // Specify the target ADS device to talk to, by NetID and AMS port.
    // Port 851 usually refers to the first PLC instance.
    let device = client.device(ads::AmsAddr::new([169, 254, 13, 12, 1, 1].into(),851));

    // Ensure that the PLC instance is running.
    assert!(device.get_state()?.0 == ads::AdsState::Run);

    // Request a handle to a named symbol in the PLC instance.
    let handle = Handle::new(device, "MY_SYMBOL")?;

    // Read data in form of an u32 from the handle.
    let value: u32 = handle.read_value()?;
    println!("MY_SYMBOL value is {}", value);

    // Connection will be closed when the client is dropped.
    Ok(())
}`

[169, 254, 13, 12, 1, 1] is the id.

But when running it, I got the following error:
Error: Ads("read state", "Port disabled - system service not started", 18)

But when I run the python code using pyads, I can see the correct result.
here's the source code:
`import pyads

# connect to plc and open connection
plc = pyads.Connection('169.254.13.12.1.1', pyads.PORT_TC3PLC1)
plc.open()

# plc = pyads.Connection('169.254.13.12.1.1', 851, '127.0.0.1')
# plc.open()

print(f"opened?:{plc.is_open}")
print(f"local address?: {plc.get_local_address()}")
# read int value by name
i = plc.read_by_name("GVL.int_val")
print(i)
# write int value by name
plc.write_by_name("GVL.int_val", i)

# close connection
plc.close()`

and the result is:
`opened?:True
local address?: <AmsAddress 169.254.13.12.1.1:33492>`

I have searched both the Beckhoff website and google the possible solutions before posting this issue, but still cannot have it resolved.

Can anyone help to advise the cause and how to address it?

yichaoliang, 2024-10-15:
I tried to troubleshoot. Then I found the following script in the source code.
```
pub enum Source {
    /// Auto-generate a source address from the local address and a random port.
    Auto,
    /// Use a specified source address.
    Addr(AmsAddr),
    /// Request to open a port in the connected router and get the address from
    /// it.  This is necessary when connecting to a local PLC on `127.0.0.1`.
    Request,
}
```
Because I run the TwinCAT software on the same machine as this script,
So I change the Source para from "Auto" to "Request".
```
let client = ads::Client::new(("127.0.0.1", ads::PORT), ads::Timeouts::none(),
                                  ads::Source::Request)?;
```
Then I ran the code, but still I am getting the following error message. 
Appreciate if anyone can help to troubleshoot.
```
Error: Ads("read state", "Target port not found, possibly ADS server not started", 6)
```

birkenfeld, 2024-10-15:
Thanks for the detailed report! Unfortunately it's hard for me to reproduce. Is it possible that you could record traces (using Wireshark or something similar) of what pyads is doing compared to ads-rs?

birkenfeld, 2024-10-15:
I also noticed, in the output of pyads you just gave
```
opened?:True
local address?: <AmsAddress 169.254.13.12.1.1:33492>
```
but not from the `read_by_name` call. Are you sure the pyads code runs completely through?

The ads-rs code will probably also connect fine and have a local address, but the error appears during the `assert!(device.get_state()?.0 == ads::AdsState::Run);` call.

yichaoliang, 2024-10-15:
``> Request

thanks for the reply.

I don't know how I should start with the wireshark to trace the message, I am still quite new to Rust.

for the client::new function, should I use Source::Auto or Source::Request?

I also tried to use another rust crate, "rust_ads_client". It works.

Compare the two crates, I can see that the "Client::new" function is a bit different and the main difference is the "route" setting.

here's the source code:

```
#![allow(unused_imports)]
use rust_ads_client::client::Client;
use ads_proto::{
    error::AdsError,
    proto::ams_address::{AmsAddress, AmsNetId},
};
use std::net::Ipv4Addr;
use std::time::Instant;

fn main() {
    let before = Instant::now();
    //Create client. If route = None then targed is local machine
    let ams_address = AmsAddress::new(AmsNetId::new(169, 254, 13, 12, 1, 1), 851);
    //let ipv4 = Ipv4Addr::new(127, 0, 0, 1);
    //let mut client = Client::new(ams_address, Some(ipv4));
    let mut client = Client::new(ams_address, None);
    //Connect client
    client.connect().expect("Failed to connect!");

    //var name and length
    let var = "GVL.int_val";
    let len = 2;

    //read data by name
    let iterations = 200;
    println!("Read var {:?} {:?} times", var, iterations);
    for _ in 0..iterations {
        match client.read_by_name(var, len) {
            Ok(r) => {
                // println!("{:?}", r);
            }
            Err(e) => {
                if e.is::<AdsError>() {
                    if let Some(e) = e.downcast_ref::<AdsError>() {
                        println!("Ads Error{:?}", e);
                        if client.connect().is_ok() {
                            println!("Reconnected...");
                        } else {
                            println!("Reconnecting failed...");
                        }
                    }
                }
                println!("Other Error{:?}", e);
            }
        }
    }
    println!("Elapsed time: {:.2?}", before.elapsed());

    //Release handle if not needed anymore
    let result = client.release_handle(var);
    println!("{:?}", result.unwrap());
}
```

yichaoliang, 2024-10-15:
maybe a side question, once I am able to read one value from the ads, the next step is to read a data structure.
For example, I have a DUT
```
TYPE udtDI :
STRUCT
    inDisc : BOOL; // Discrete input
    cfgInvertInput : BOOL; // Invert the input signal (1: invert enable)
END_STRUCT
END_TYPE
```

And I have a instance LSL001 of udtDI.

I want to read the two values from LSL001, can I read them by sum_read or by multi_read? And is Multi_Read faster than reading the tags one by one?

yichaoliang, 2024-10-15:
> I also noticed, in the output of pyads you just gave
>
> ```
> opened?:True
> local address?: <AmsAddress 169.254.13.12.1.1:33492>
> ```
>
> but not from the `read_by_name` call. Are you sure the pyads code [...]

here's the updated code and execution result:
code:
```
import pyads
import time

t0 = time.time()

# connect to plc and open connection
plc = pyads.Connection('169.254.13.12.1.1', pyads.PORT_TC3PLC1)
plc.open()

# plc = pyads.Connection('169.254.13.12.1.1', 851, '127.0.0.1')
# plc.open()

print(f"opened?:{plc.is_open}")
print(f"local address?: {plc.get_local_address()}")
# read int value by name
for j in range(200):
    i = plc.read_by_name("GVL.int_val")
print(i)
# write int value by name
# plc.write_by_name("GVL.int_val", i)

# close connection
plc.close()

t1 = time.time()

total = t1-t0
print(f"execute time:",total)
```

result:
```
opened?:True
local address?: <AmsAddress 169.254.13.12.1.1:33634>
False
execute time: 0.638174295425415
```

yichaoliang, 2024-10-15:
> Thanks for the detailed report! Unfortunately it's hard for me to reproduce. Is it possible that you could record traces (using Wireshark or something similar) of what pyads is doing compared to ads-rs?

I restarted the TwinCAT on my laptop, now I can see the connection. Weird.

Now I can read the value, I need to change the ads::Source to "Request".

Here's the source code:

```
#![allow(unused_imports)]
use ads::client::Client;
use ads::Handle;
// use roboplc::comm::Timeouts;
// use roboplc_io_ads as ads;
use std::time::Duration;
fn main() -> ads::Result<()> {
    // let ipv4 = Ipv4Addr::new(127, 0, 0, 1);
    // Open a connection to an ADS device identified by hostname/IP and port.
    // For TwinCAT devices, a route must be set to allow the client to connect.
    // The source AMS address is automatically generated from the local IP,
    // but can be explicitly specified as the third argument.
    let client = ads::Client::new(("127.0.0.1", ads::PORT), ads::Timeouts::none(),
                                  ads::Source::Request)?;

    // Specify the target ADS device to talk to, by NetID and AMS port.
    // Port 851 usually refers to the first PLC instance.
    let device = client.device(ads::AmsAddr::new([169, 254, 13, 12, 1, 1].into(), 851));

    // Ensure that the PLC instance is running.
    assert!(device.get_state()?.0 == ads::AdsState::Run);

    // Request a handle to a named symbol in the PLC instance.
    let handle = Handle::new(device, "GVL.int_val")?;

    // Read data in form of an u32 from the handle.
    let value: u32 = handle.read_value()?;
    println!("MY_SYMBOL value is {}", value);

    // Connection will be closed when the client is dropped.
    Ok(())
}
```

And here's the result:
```
MY_SYMBOL value is 22
```

birkenfeld, 2024-10-15:
Ok, great :) I'll have a look at your other question soon.

yichaoliang, 2024-10-16:
> Ok, great :) I'll have a look at your other question soon.

Thanks, I will close this current ticket and open a new one for the separate issue.

yichaoliang, 2024-10-16:
ticket closed.

closed 2024-10-16 by yichaoliang