birkenfeld/ads-rs #30: how to read instance of data structure from the ads

yichaoliang, 2024-10-16:
For example, I have a DUT
```
TYPE udtDI :
STRUCT
    inDisc : BOOL; // Discrete input
    cfgInvertInput : INT; // Invert the input signal (1: invert enable)
END_STRUCT
END_TYPE
```

And I have a instance LSL001 of udtDI.
so it has two elements.
LSL001.inDisc
LSL001.cfgInvertInput

The two elements are different data type.

I want to read LSL001 as a lot, can I read the values by sum_read or by multi_read? And is Multi_Read faster than reading the tags one by one?

yichaoliang, 2024-10-16:
I found this post, similar question, but to pyads.
It's a good reference to my question here.
https://stackoverflow.com/questions/74134828/how-to-access-a-nested-structure-using-pyads

birkenfeld, 2024-10-17:
You can define a struct in Rust and derive the relevant `zerocopy` traits. I just pushed an example: 
https://github.com/birkenfeld/ads-rs/blob/master/examples/values_via_handle.rs

closed 2024-11-09 by birkenfeld