birkenfeld/ads-rs #31: Connecting to ADS from Tauri rust backend

EFeru, 2024-11-08:
Dear @birkenfeld ,

I would like to use your library to connect to a local PLC using ADS protocol within the [Tauri](https://v2.tauri.app/start/) rust backend. However, I have issues integrating the example you provided (also because I am lacking Rust knowledge).

Question:
How do you suggest integrating the example in Tauri? Any help is highly appreciated.

These are the Tauri rust files:
`main.rs`
```rs
// Prevents additional console window on Windows in release, DO NOT REMOVE!!
#![cfg_attr(not(debug_assertions), windows_subsystem = "windows")]
fn main() {
    app_lib::run();
}
```

`lib.rs`
```rs
#[cfg_attr(mobile, tauri::mobile_entry_point)]
use tauri::Manager;
pub fn run() {
    tauri::Builder::default()
        .plugin(tauri_plugin_process::init())
        .plugin(tauri_plugin_updater::Builder::new().build())
        .setup(|app| {
            if cfg!(debug_assertions) {
                app.handle().plugin(
                    tauri_plugin_log::Builder::default()
                        .level(log::LevelFilter::Info)
                        .build(),
                )?;
            }

            Ok(())
        })
        .run(tauri::generate_context!())
        .expect("error while running tauri application");
}
```

What I did already?
I tried directly like this in `lib.rs`:
```rs
#[cfg_attr(mobile, tauri::mobile_entry_point)]
use tauri::Manager;
pub fn run() {
    tauri::Builder::default()
        .plugin(tauri_plugin_process::init())
        .plugin(tauri_plugin_updater::Builder::new().build())
        .setup(|app| {
            if cfg!(debug_assertions) {
                app.handle().plugin(
                    tauri_plugin_log::Builder::default()
                        .level(log::LevelFilter::Info)
                        .build(),
                )?;
            }

            // -----------------------
            let client = ads::Client::new(("localhost", ads::PORT), ads::Timeouts::none(), ads::Source::Auto).unwrap();
            let device = client.device(ads::AmsAddr::new([192, 168, 56, 1, 1, 1].into(), 851));
            // assert!(device.get_state()?.0 == ads::AdsState::Run);
            let handle = ads::Handle::new(device, "MAIN.PLC_Counter").unwrap();
            let value: i32 = handle.read_value().unwrap();
            println!("MY_SYMBOL value is {}", value);
            // -----------------------
            Ok(())
        })
        .run(tauri::generate_context!())
        .expect("error while running tauri application");
}
```
However, I get a crash
```
thread 'main' panicked at src\lib.rs:17:111:
called `Result::unwrap()` on an `Err` value: Io("connecting TCP socket", Os { code: 10061, kind: ConnectionRefused, message: "No connection could be made because the target machine actively refused it." })       
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[1108/175206.927:ERROR:window_impl.cc(121)] Failed to unregister class Chrome_WidgetWin_0. Error = 1412
```

birkenfeld, 2024-11-08:
Hi! First of all, this looks unrelated to Tauri. I assume that you get the same error when running the example standalone. I can't give much advice about Tauri since I don't know it.

Since the PLC is local, this is probably Windows. My first suggestion would be to use "127.0.0.1" instead of "localhost", I seem to remember some weirdness on Windows there. 

If that makes no difference, are you sure TwinCAT is running on the same host?

Also, you should use Source::Request but that only makes a difference when a connection is established in the first place. I need to update that example. 

Finally, you can maybe try an ADS library in a language that you're more familiar with, if that works but this library doesn't we have something to compare.

EFeru, 2024-11-08:
Hi,

 I am indeed using windows. I did try using another package for node.js, more exactly https://github.com/jisotalo/ads-client
with which I managed to connect via ADS with the PLC by only specifying the 2 parameters
targetAmsNetId: '192.168.65.1.1.1',
targetAdsPort: 851,

I will try tomorrow your suggestion and see if I get further and let you know.
Thanks for your response.

EFeru, 2024-11-09:
Hi @birkenfeld ,

I made a complete separate example `test.rs`:
```rs
fn main() -> ads::Result<()> {
    // Open a connection to an ADS device identified by hostname/IP and port.
    // For TwinCAT devices, a route must be set to allow the client to connect.
    // The source AMS address is automatically generated from the local IP,
    // but can be explicitly specified as the third argument.
    let client = ads::Client::new(("127.0.0.1", ads::PORT), ads::Timeouts::none(),
                                  ads::Source::Auto)?;

    // Specify the target ADS device to talk to, by NetID and AMS port.
    // Port 851 usually refers to the first PLC instance.
    let device = client.device(ads::AmsAddr::new([192, 168, 56, 1, 1, 1].into(), 851));

    // Ensure that the PLC instance is running.
    assert!(device.get_state()?.0 == ads::AdsState::Run);

    // Request a handle to a named symbol in the PLC instance.
    let handle = ads::Handle::new(device, "MAIN.PLC_Counter")?;

    // Read data in form of an u32 from the handle.
    let value: i32 = handle.read_value()?;
    println!("MY_SYMBOL value is {}", value);

    // Connection will be closed when the client is dropped.
    Ok(())
}
```
And I run it with carg `cargo run --bin test`. The result:
```
   Compiling app
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.99s
     Running `target\debug\test.exe`
Error: Ads("read state", "Port disabled - system service not started", 18)
error: process didn't exit successfully: `target\debug\test.exe` (exit code: 1)
```

However, I know the ADS is working since I can connect and read values when using [node,js ADS library](https://github.com/jisotalo/ads-client)

Any suggestions on what to do next?

birkenfeld, 2024-11-09:
Ok, this looks better. First, please use `ads::Source::Request` when connecting to the local TwinCAT.

Otherwise you can try to omit the `assert!(device.get_state()?.0 == ads::AdsState::Run);` line and see if reading symbols works.

In general this looks very similar to #29 which was solved, apparently, by a TwinCAT restart.

If nothing helps, then I would be interested in a wireshark trace of the JS library vs this library :)

EFeru, 2024-11-09:
Yes, indeed this looks like #29. I changed to `ads::Source::Request` and it works. Nice!
Full example:
```rs
fn main() -> ads::Result<()> {
    let client = ads::Client::new(("127.0.0.1", ads::PORT), ads::Timeouts::none(),
                                  ads::Source::Request)?;
    let device = client.device(ads::AmsAddr::new([192, 168, 56, 1, 1, 1].into(), 851));
    assert!(device.get_state()?.0 == ads::AdsState::Run);
    let handle = ads::Handle::new(device, "MAIN.PLC_Counter")?;

    let value: i32 = handle.read_value()?;
    println!("MY_SYMBOL value is {}", value);

    Ok(())
}
```

Result:
```
MY_SYMBOL value is 23
```

Thanks for your great help. 

Now I will look into how can I move all the initalization
```rs
    let client = ads::Client::new(("127.0.0.1", ads::PORT), ads::Timeouts::none(),
                                  ads::Source::Request)?;
    let device = client.device(ads::AmsAddr::new([192, 168, 56, 1, 1, 1].into(), 851));
    assert!(device.get_state()?.0 == ads::AdsState::Run);
    let handle = ads::Handle::new(device, "MAIN.PLC_Counter")?;
```
into an init function and make another function to retrieve the value. I under-estimate Rust, does not seems very intuitive.
If you have any suggestion on how to do it (ChatGPT does not do very well in Rust) let me know. 
I wait for your answer and then I close the issue.

birkenfeld, 2024-11-10:
Nice! The problem here is that `device` and `handle` wrap the `client`, carrying the reference to it around as a lifetime. This can be a bit tricky to integrate, and how to do it depends on the concrete circumstances of your application. I'd suggest you ask about it in a general Rust help forum, maybe on Reddit, Discuss or Discord. But thanks for the detailed issue, and feedback!

I'll update the examples and README to better reflect the need to use `Source::Request` for localhost.

closed 2024-11-10 by birkenfeld