birkenfeld/ads-rs #38: Read multi question

albeporta, 2025-04-08:
goodmorning,
i have a problem using the function read_multi; i'm trying with example found in doc but i have an error with rust checked

`  
let mut buf_1 = [0; 128]; // request reading 128 bytes
let mut buf_2 = [0; 128]; // from two indices

// create the request structures
let mut req_1 = ReadRequest::new(sym1.group, sym1.offset, &mut buf_1);
let mut req_2 = ReadRequest::new(sym2.group, sym2.offset, &mut buf_2);

//  actual request
device.read_multi(&mut [req_1, req_2])?;

// extract the resulting data, checking individual reads for
// errors and getting the returned data otherwise
let res_1 = req_1.data()?;
let res_2 = `req_2.data()?;
`

rust compiler give this error

400 |     let mut req_2 = ReadRequest::new(sym2.group, sym2.offset, &mut...
    |         --------- move occurs because `req_2` has type `ReadRequest<'_>`, which does not implement the `Copy` trait
401 |     //  actual request
402 |     device.read_multi(&mut [req_1, req_2])?;
    |                                    ----- value moved here
...
406 |     let res_2 = req_2.data()?;
    |                 ^^^^^ value borrowed here after move

same for req_1.

Can you help me ?
thanks 
Alberto

birkenfeld, 2025-04-09:
Hi Alberto,

sure! This is a bit tricky due to the restrictive handling of Rust's mutable references. Basically you're moving the requests into the temporary array `[req_1, req_2]` which you have then no way of retrieving. Instead, you can create the array beforehand and reference it:
```
    let mut buf_1 = [0; 128]; // request reading 128 bytes
    let mut buf_2 = [0; 128]; // from two indices

    // create the request structures
    let req_1 = ReadRequest::new(sym1.group, sym1.offset, &mut buf_1);
    let req_2 = ReadRequest::new(sym2.group, sym2.offset, &mut buf_2);
    let mut requests = [req_1, req_2];

    //  actual request
    dev.read_multi(&mut requests).unwrap();

    // extract the resulting data, checking individual reads for
    // errors and getting the returned data otherwise
    let [req_1, req_2] = requests;
    let res_1 = req_1.data();
    let res_2 = req_2.data();
```

birkenfeld, 2025-04-09:
I will fix the documentation accordingly.

closed 2025-04-09 by birkenfeld

albeporta, 2025-04-10:
thanks.
bye Alberto

commit a17aa9b2 (2025-04-09): client: fix docstring for `read_multi`, compile (but not run) doctests

Fixes #38